ms-iot/rpi-iotcore #21: Build errors for 10.0.10586.0 missing header files

ErwinZwart, 2016-04-14:
Installed the latest IoTCore environment but the solution does not compile.:

first error:
\bsp\drivers\audio\bcm2836\rpiwav.h(12): fatal error C1083: Cannot open include file: 'portcls.h': No such file or directory

The include path mentions:
C:\Program Files (x86)\Windows Kits\10\Include\10.0.10586.0\shared\

all the missing header files are located in a previous version of the Windows 10 kit e.g.
C:\Program Files (x86)\Windows Kits\10\Include\10.0.10240.0\km

The km folder is missing in version 10.0.10586.0
Am I missing something

douglascgh, 2016-04-14:
The missing km folder is suspicious, portcls.h can be found at "C:\Program Files (x86)\Windows Kits\10\Include\10.0.10586.0\km". You could repair/re-install your WDK installation so the headers for 10586 are correctly placed.

Alternatively you can also update your SDK and WDK to the current latest insider version to see if that helps with your setup.

VS2015 Update 2 : http://go.microsoft.com/fwlink/p/?LinkId=534599
SDK 14295 : http://go.microsoft.com/fwlink/?LinkId=615097
WDK 14295 : http://go.microsoft.com/fwlink/p/?LinkId=760768

ErwinZwart, 2016-04-17:
Issue resolved by installing the latest WDK and SDK!

closed 2016-04-17 by ErwinZwart