ms-iot/rpi-iotcore #39: Error: No package definition files found.

CharlieTao, 2018-05-21:
Download and Unzip binaries,Run build.cmd,Can't build package.
Prompt "Error: No package definition files found."

CharlieTao, 2018-05-22:
Run buildpkg .pkg.xml will generate wm.xml,but when Processing wm.xml,There's another error. (PkgGenCommon) : error : Failed DoConvert

CharlieTao, 2018-05-22:
Modify .inf file can buile suceess.

[Manufacturer]                                                     [Manufacturer]
%MSFT%=Standard,ntarm     ------------>         %MSFT%=Standard,ntx86
[Standard.ntarm]                                                  [Standard.ntx86]

saraclay, 2018-09-13:
Hi @CharlieTao, this may have been fixed in recent updates so I will close this for now. If that is not the case, please file an issue here: https://github.com/MicrosoftDocs/windows-iotcore-docs

closed 2018-09-13 by saraclay