ms-iot/rpi-iotcore #41: Update to match recent official and Insider builds, add Model B+ drivers

Googulator, 2018-06-06:
Recent official release and Insider builds for the Raspberry Pi appear to come with never drivers than what's available here, including some completely new drivers (e.g. USB).

In addition, an Insider build is now available for the Raspberry Pi 3 Model B+, bringing additional drivers for B+-specific devices.

Please update this repository with the new content found in recent builds.

saraclay, 2018-10-09:
Hello, new content is being updated here: https://github.com/MicrosoftDocs/windows-iotcore-docs

closed 2018-10-09 by saraclay

Googulator, 2018-10-09:
That repository contains documentation only. What's missing is the updated drivers provided in newer IoT Core builds themselves.