ms-iot/rpi-iotcore #44: Raspberry PI 3 Model B - Windows 10 IoT leaves GPIO Pin to a different state than Raspbian

alexiordan, 2018-09-29:
Given 2 pieces of code that should behave the same, Windows 10 IoT (v. 10.0.17134.1) outcome is different from Raspbian one:

I am not sure which one behaves correclty. Please advise.

Pin under test is GPIO14 that has an internall pull-down resistor.

**Code 1 - Python - Raspbian**
```
RESET_WATCHDOG1 =  14
def resetWatchDog():
   GPIO.setup(RESET_WATCHDOG1, GPIO.OUT)
   GPIO.output( RESET_WATCHDOG1, False)
   time.sleep(0.500)
   GPIO.setup(RESET_WATCHDOG1, GPIO.IN)

while True:
    resetWatchDog()
    time.sleep(10)
```
Read more: http://forum.switchdoc.com/thread/444/watchdog-reboots-system-4-seconds#ixzz5SWcd2Amx

**Code 2 - .Net Core - Windows 10 IoT**
```
///called each 10 seconds
private async Task PatTheDog()
{
      m_WatchdogPetPin.SetDriveMode(GpioPinDriveMode.Output);
      m_WatchdogPetPin.Write(GpioPinValue.Low);
      await Task.Delay(500);
      m_WatchdogPetPin.SetDriveMode(GpioPinDriveMode.Input);
}
```

Folowing happens:
Raspbian
- When pin is set LOW Voltage goes to 0
- When pin is set as INPUT, Voltage increases slowly up to ~0.7 (then another cycle comes and is set to 0)
- When the loop ends Voltage grows slowly up to 3.3V

Windows 10 IoT
- When pin is set LOW Voltage goes to 0
- When pin is set as INPUT, Voltage increases slowly up to ~0.7 (then another cycle comes and is set to 0)
- **When the loop ends Voltage grows up to ~0.7 and stays there. It is not growing up to 3.3V...**

I tried also setting:
`m_WatchdogPetPin.SetDriveMode(GpioPinDriveMode.InputPullUp);`
but then 
- When pin is set as INPUT, Voltage increases slowly up to 3.3V

jordanrh1, 2018-10-11:
This is the expected behavior. When you set the pin to `GpioPinDriveMode.Input` it will enter a high impedance state, which means it will float based on capacitance, leakage, etc... It could take on any voltage depending what it's attached to. 

Is the pin attached to something?

parameshbabu, 2018-11-16:
Closing this as the query is answered.

closed 2018-11-16 by parameshbabu