ms-iot/rpi-iotcore #55: Update Board Support Package to support Raspberry Pi 4

sr229, 2019-12-03:
Raspberry Pi 4 uses a newer SoC compared to Raspberry Pi 3, so you'll need to write a new BSP for this chipset.

crpietschmann, 2019-12-08:
it looks like this project is no longer maintained by Microsoft according to the main Readme.md.

    > "This BSP repository is under community support"

We may never see Raspberry Pi 4 support for Windows 10 IoT Core, unfortunately...

sr229, 2019-12-09:
@crpietschmann I might give this a shot since I am a madman and I like a challenge.

Do we have documentation to port BSPs?

minecraftchest1, 2019-12-31:
Question1: what is a BSP?
Question2: why doen't we just look at the raspberrian source and build the BSP ourselves?

sr229, 2020-01-02:
@minecraftchest1 BSP is short for "Board Support Package", an exclusive package to run Windows on ARM devices. These usually contain drivers

And sadly, I don't have access to source code for RPi Firmware, so we have to eyeball it using the documentation. Drivers are written in pure C, please refer to the Raspberry Pi 3 BSP.

minecraftchest1, 2020-01-02:
Look under github.com/raspberrypi for the source.

sr229, 2020-01-02:
@minecraftchest1 Like I said, Firmware is **closed** source. The only thing we can eyeball for drivers is the **documentation**.

I already know the repo, no need to link it :/

minecraftchest1, 2020-01-02:
I just looked at the rep. and read the description. My apologies.

minecraftchest1, 2020-01-02:
What do we need to migrate IoT core to raspberryp4 though?

minecraftchest1, 2020-01-02:
Could we use some of the code from userland?

sr229, 2020-01-02:
> Could we use some of the code from userland?

No.

galvesribeiro, 2020-01-03:
The *firmware* is closed source but it is made by RPI foundation/company, right? If that is truth, theoretically they can release the firmware for the updated board and the drivers for Windows 10 IoT that comes from this repo can be follow after it, right?

As far as I know, you can customize Windows 10 IoT and create custom images using different BSPs as long as the OS arch is the same as the CPU underneath... 

Can anyone confirm that?

minecraftchest1, 2020-01-03:
I was thinking we can see if we can get the firmware binaries to work with the BSP

Sent from my T-Mobile 4G LTE Device

5he11, 2020-01-04:
@minecraftchest1 I think the firmware binaries wouldn't work because we will need to port it to be based on Windows Driver Foundation (WDF). That's how drivers work on Windows including this Windows 10 IoT Core version.
so basically what we will need to do to make Windows 10 IoT Core to support RPi 4 is to take a look and see what chips are used by RPi 4 and we modify their Linux drivers to be compatible with WDK, compile, pack them into BSP, build Windows 10 IoT Core image with this new BSP.

KenD123, 2020-01-27:
@sr229 
Are you still interested in developing a BSP for w10-iotcore to run on Pi4?  I'm not an accomplished coder but I'm pretty good at sniffing out important information.  Anybody, please let me know if you're still interested.
Thanks

sr229, 2020-01-28:
I'm still reading the documentation, its a bit of a stretch to get it working at this moment

KenD123, 2020-01-28:
That's great.  I'm not expecting you to do it all by yourself.  I was actually hoping  to learn something by collaborating on this worth while project. It's not at all about what we do or don't know at the moment, if it was easy it would have been done long ago.  If you have time, I think we should start with something relatively easy. Perhaps getting the Wifi and bluetooth drivers working in the Pi3B+.
In the current version there doesn't seem to be any Lan or Bluetooth drivers in the open source BSP, but since it works on the Pi3B I assume that older driver in the closed source operating system.  SO the task at hand seems to be, Identify the new hardware on the B+ for the wifi Bluetooth.  I think its a broadcom part.  Then finding out exactly which drivers the linix system used to address the new hardware,hopefully we find the source in C or C++.  Then the fun part. figuring out how to take that source and turn it into a functioning driver for the IOTcore OS. Thoughts, suggestions,Ideas?Thanks

minecraftchest1, 2020-01-28:
I agree.

Sent from my T-Mobile 4G LTE Device

KenD123, 2020-01-29:
From:https://blog.pimoroni.com/a-deep-dive-into-the-raspberry-pi-3-b-hardware/

The chip, that takes up most of the space underneath that metal shielding, is the Cypress CYW43455, with faster 802.11ac dual-band (2.4 and 5GHz) wireless LAN and Bluetooth 4.2. The Pi 3 used the Cypress CYW43438 chip, and 802.11n 2.4GHz wireless chip, as does the Pi Zero W.

3B appears 2016 while 3B+was introduced in 2018.

KenD123, 2020-01-30:
Also from the link above:
The Pis 2 and 3 used the SMSC LAN9514 chip, a USB 2.0 and 10/100 ethernet chip, whereas the Pi 3 B+ uses a Microchip LAN7515 chip, theoretically capable of gigabit speeds, although limited by the speed of the USB 2.0 bus over which it runs to the SoC.

KenD123, 2020-01-30:
https://downloads.raspberrypi.org/raspbian/release_notes.txt

This seams to indicate Pi3b+ w/43455 support was released 3/13/2018
Prior release was 11/29/2017

nithinmanne, 2020-01-31:
Is this still being worked on? I would love to contribute if there's something I can do. I have some experience porting drivers to WDK, as I worked on porting Snapdragon BSP to WOA for the SQ1 Surface.

galvesribeiro, 2020-01-31:
@nithinmanne please ping me on gutemberg at outlook.com at skyper or email. I may have something for you offline. Thanks

KenD123, 2020-01-31:
@nithinmanne 
You sound like just what we need.  What do you need next? I was working on comparing the raspbian OS from when the wifi worked Pi3B and when it worked with Pi3B+, hoping that the changes were modest and that we could get wifi to work on w10iot.  So far the changes seem non trivial, and I don't think the original wifi drivers were part of the BSP for W10iot, so we need a driver from scratch. If you can point me in a better direction, I'm on board.

sr229, 2020-02-02:
I started to fork the Repository so I can begin the preliminary work on the drivers. If anyone's willing to help let me know.

What I've discovered so far is: 

- CPU and GPU interface wise, there has been no difference according to documentation, so we might just redefine them so it points properly for the new Broadcom chip.

- WiFi/BT is something I'll investigate since RPi has made no documentation for those other than the Mailbox interface they introduced.

KenD123, 2020-02-03:
https://community.cypress.com/servlet/JiveServlet/download/14597-7-34098/cypress-fmac-v4.12-2018_0205.zip 

This zip archive has a readme with additional information about the linux driver.  I'm trying to get my head around this.  Do the linux drivers leverage the driver that was provided for the linux platform from the silicone manufacturer? If they do, it seems like we have a long row to hoe, as the Manufacturer has no plans to provide "Drivers" for windows 10 or windows 10 IOT.

sr229, 2020-02-04:
@KenD123 I think we can ask the @raspberrypi people to help us with this since this requires direct support with them.

KenD123, 2020-02-04:
from https://www.raspberrypi.org/forums/viewtopic.php?f=105&t=259249
(last post)

"I'm Trading rather than Foundation, but as far as I know, we do not intend to do a BSP for Win10 for the Pi4. Apart from a lack of any real profit motive, we have limited staff available and none with Win10IoT experience, and what staff we do have are pretty busy on more important stuff. We would probably be happy supporting MS with advice if they wanted to do it, but that would be the limit of our involvement.

I suspect that a BSP could be written by a third party experienced in it, using any existing one from earlier versions, and with help from the forums and the Linux source code."

KenD123, 2020-02-05:
I wonder if we should start a campaign to ask Raspberrypi to ask Cypress to develop drivers for W10 IOT.

KenD123, 2020-02-11:
I can't get the repository to compile in VS2019 community.  I keep getting 17 errors (Spectre-migrated libraries are required) but I feel like Ive added every library that it could be asking about. Can somebody hold my hand for a minute? Thanks

KenD123, 2020-02-11:
This is what I've been trying
https://github.com/ms-iot/rpi-iotcore/pull/54/commits/62e5d975017e30bf9422cc2a1abce4207b848ebe

nithinmanne, 2020-02-11:
@KenD123  I was able to build without issues, but I also pulled changes from that pull request(https://github.com/ms-iot/rpi-iotcore/pull/54) whose commit you mentioned. I installed all toolchains for ARM, and also the latest WDK.

KenD123, 2020-02-12:
Thanks nithinmanne, I'm not quite there but I'm a lot closer. I needed these two "individual components
"  as well .https://blog.csdn.net/qq_34842360/article/details/100764245. Now I have missing include directories.  I'm not sure how to "pull" changes but I bet if I could it would compile.

nithinmanne, 2020-02-12:
Yeah, thats the issue I faced. Clone from https://github.com/christopherco/rpi-iotcore/tree/chrco/rs5-update and checkout that specific tree "chrco/rs5-update"

KenD123, 2020-02-12:
60 warnings but it's compiling completely! Thanks again.  Next step is to try to load it into the Pi.

sr229, 2020-02-13:
Looks like we're seeing progress, I'm happy to help out but my studies are getting in the way at the moment.

Marcinoo97, 2020-02-16:
BSP for Rpi4 would be the best thing ever!

jdormand, 2020-03-07:
Hi - slight abuse of this thread but I'm desperately looking for help on a Pi/IoTCore project. This query is semi-related to this Pi4 BSP topic, and I would happily contribute to the project if there is anything I can do.

My immediate query relates to IoTCore on a CM3 compute module. Given the GPIO 28-45 are unassigned in hardware on the CM, I was hoping a simple bit of ASL/RhProxy work would make them all available to user land.

I have followed MS instructions here to the letter: https://docs.microsoft.com/en-us/windows/uwp/devices-sensors/enable-usermode-access but not joy. My simple ASL that enables GPIO30 and 31 only, compiles no problem, however there is no evidence that the resulting ACPITBL.DAT is being loaded. And the gitbug sourced GpioTestTool.exe does not show any GPIO above 27 being available.

Any pointers? Sorry again for hijacking the thread :/

mklemarczyk, 2020-03-17:
Hello, I can handle testing if you have any version to verify.
Can we share the solution to work on it together? Some branch or fork?

I would like to help extend support to Raspberry Pi 4 and Paspberry Pi Zero W.

andreiw, 2020-03-25:
Folks... see https://rpi4-uefi.dev/. The Pi 3 64-bit UEFI is now part of edk2 platforms (tiano core) upstream and it supports WoA. Now we are extending this code to Pi 4 and can boot Windows PE or installer

andreiw, 2020-03-25:
You might remember RaspberryPiPkg - this is fully integrated as  https://github.com/tianocore/edk2-platforms/tree/master/Platform/RaspberryPi/RPi3 and there is https://github.com/tianocore/edk2-platforms/tree/master/Platform/RaspberryPi/RPi4 now

driver1998, 2020-03-27:
With [64-Bit TianoCore UEFI](https://github.com/tianocore/edk2-platforms/tree/master/Platform/RaspberryPi/RPi4), [my ARM64 fork of the BSP](https://github.com/driver1998/bsp), and [the non-commercial ARM64 release of the RPi USB driver](https://pi64.win/download-usb-drivers/) (used in USB-C OTG port on the Pi4) , it should be possible to get at least some feature to work on Pi4.

Although these efforts are focused on Windows 10 Desktop (and PE) on ARM64, so some fiddling might be needed for these to work with IoT Core.

EddieDemon, 2020-06-15:
@nithinmanne @KenD123 @sr229 Since neither of you publicly shared the repository we're all unable to get updates or contribute to this cause. Please share your current information.

sr229, 2020-06-15:
> @nithinmanne @KenD123 @sr229 Since neither of you publicly shared the repository we're all unable to get updates or contribute to this cause. Please share your current information.

I'm not obligated to share my progress since I'm doing this on my free time. It comes out when its ready.

EddieDemon, 2020-06-15:
Okay, got that clear. We'll await your feedback.

sr229, 2020-06-15:
> With [64-Bit TianoCore UEFI](https://github.com/tianocore/edk2-platforms/tree/master/Platform/RaspberryPi/RPi4), [my ARM64 fork of the BSP](https://github.com/driver1998/bsp), and [the [...]

Honestly I want to see Windows 10 Desktop working first before IoT Core since having a successful working Windows 10 setup would reflect that we're doing everything right.

driver1998, 2020-08-19:
WOW this is great news, XHCI on >3GB RAM and GENET are both big issues in the way of Windows 10 on Raspberry Pi 4.
Hopefully #64 and #63 can both be merged soon.